Asuswrt-Merlin/asuswrt-merlin.legacy #1027: Add 100+% Transmit Power, Add Guests to Clients View List

schwab1, 2016-09-05:
I own an Asus RT-AC66u and an Asus RT-AC68U.

I installed the AsusWRT-Merlin 380.61 firmware.

While I am not a programmer I have two enhancements to suggest.

(1) Please rewrite/copy the  firmware to revert back to the factory option where users can **increase the Transmit Power(Performance) above 100.**

(2) From the Network Map under 'Clients' and 'View List' write the firmware to **designate 'Guest Clients' logged in** under the All and/or Interface tabs.

I hope these suggestions may be considered in future releases.

Thank you for your consideration.

RMerl, 2016-09-05:
Power output cannot be changed, both for legal reasons and technical reasons (the code is now closed source, to prevent legal issues).

Asus is constantly changing the Networkmap code, so I prefer not to make any change to it, or I would have to constantly rewrite any changes I make.

closed 2016-09-05 by RMerl

schwab1, 2016-09-05:
Does it make sense to revert my firmware back to a legacy version in order
to increase transmit power >100?

What version might I use?

Advantages vs Disadvantages choosing legacy firmware?

Thanks for your reply to my original question.

Tom