Asuswrt-Merlin/asuswrt-merlin.legacy #1028: Crsf vulnerability

baitnfatty, 2016-09-06:
Merlin, I did not see anything about this being patched in notes anywhere and do not think that the standard asus firmware has been patched. Could you give me some info? https://securityevaluators.com/knowledge/case_studies/routers/asus_rtn56u.php
I have a 66U and currently being fubar 

Please let me know. Thanks

RMerl, 2016-09-06:
This refers to a 3 years old firmware.  Asus has since moved to a token-based authentication, so this should have been resolved long ago.

baitnfatty, 2016-09-06:
my router was comped by that as well as a buffer overflow last night while I was kicking the asshats off. I know for a fact the new FW DOES NOT STOP THIS. Watched it happen with my own eyes I have the buffer overflow logged, although not the csrf. 

Booted them off while they were on it and then they over took it as soon as I reset it with new FW. Then by 3am that evening they were back in by means of buffer overflow 

FW is 380.3831 

Also, it seems like they installed their own version of the FW as every time I changed the page it would flash a page with more settings and then turn around and give me the right page.... 

I really wish I had the packets for the csrf... I'm sure it would help u more than I .  
I assume urs is most secure FW for this router va tomato and ddwrt as u have been helping them since the black knight came out.. Lol any thoughts?

baitnfatty, 2016-09-07:
FYI ill just try it and let u know, whats another days work gone ;)

baitnfatty, 2016-09-07:
actually i just found ur recommendation to solve the issues. I will let u know how it goes.

closed 2016-09-07 by baitnfatty