Asuswrt-Merlin/asuswrt-merlin.legacy #1029: Listed devices, Tx/Rx rate and the access time in Network Map menu

d3xt3r67, 2016-09-06:
I'm a new user of asuswrt-merlin and I'm pretty satisfy. But there is one issue I noticed in the NETWORK MAP menu. 

When I click on "VIEW LIST" button, the displayed informations aren't correct. 
 - If want to display all my connected devices despite the interface, none of my devices are listed 
 - If want to display all my connected devices sorted by interface, I can see my wired devices but my WIFI 
   devices aren't listed. 
 - The Tx/Rx rate and the access time doesn't works; it shows nothing
I provided some print screen 
[image]
[image]
[image]

RMerl, 2016-09-07:
Those values are shown as empty because the router thinks they are wired clients.  This usually happens if the clients are currently offline and not connected.

d3xt3r67, 2016-09-07:
All of my devices are up and running and connected straight in my router,
not behind a switch

pepeEL, 2017-01-10:
TX and RX should also show for device connected by LAN.

RMerl, 2017-01-12:
The router has no way of knowing what link rate is used by a wired client, since that client can be anywhere on your network, including behind a switch.