Asuswrt-Merlin/asuswrt-merlin.legacy #1030: RT-AC66U - 380.61 -- WPS turned on

thezoggy, 2016-09-07:
Upgraded to latest, which changed the firmware format.. upon updating I just discovered that WPS was turned on... it was previously off.

not sure if this was by design or just fallout from the firmware format change.. just noting in case others are as surprised as I was.

RMerl, 2016-09-07:
Firmware defaults to WPS being enabled if you did a factory default reset.

App icon has been there for quite some time already, it's a shortcut to Asus's iOS/Android app on their respective stores.

closed 2016-09-07 by RMerl

thezoggy, 2016-09-07:
app icon doesnt do anything for me when clicked (firefox). it wasnt there on `380.59`.. and pretty sure I didnt reset defaults unless it just did it automatically.

RMerl, 2016-09-07:
The icon popup only works on some browsers, like Chrome.