Asuswrt-Merlin/asuswrt-merlin.legacy #1036: Time management doesn't kill existing connections

apenney, 2016-09-17:
I rely on the parental controls stuff to timebox my daughter to 6 hours of Internet access a day (man, what I'd have given for 6 hours a day of internet access when I was a teen, we were stuck on a 14.4 modem with paid local calls).

I noticed the other day that if she has an established connection to Skype it doesn't disconnect when the time management stuff kicks in, it only stops new connections.  She was able to keep that connection up for an additional 3 hours, which is uncool.

Not sure if it's easily fixable, I guess you'd have to force a RST to all connections from the mac address whenever the time management system kicks in, but I thought I'd at least mention it here.

RMerl, 2016-09-17:
Disable NAT acceleration, as it causes established connections to bypass iptables.

apenney, 2016-09-18:
Will give it a go! Thanks.

closed 2016-09-26 by RMerl