Asuswrt-Merlin/asuswrt-merlin.legacy #1039: Nano cannot be compiled

hajuuk, 2016-09-25:
I'm trying to compile the newest source code, but interrupted on nano compile process.

The error message was ld: cannot find -ltinfo

If I change NANO=n, other sources are compiled correctly.

blackfuel, 2016-09-25:
I also added "# RTCONFIG_NANO is not set" to the end of ~/asuswrt-merlin/release/src/router/config_base, so that the build does not prompt the user to compile Nano, when I set "NANO=n" in target.mak.

RMerl, 2016-09-26:
The config_base entry should only be necessary if your build profile lacks any reference to NANO.  Still, I added it, in case it might happen.

RMerl, 2016-09-26:
OP: make sure you are using a supported build environment.  I don't know where this tinfo library would come from, I don't have any reference to it in my own build tree.

RMerl, 2016-10-02:
OP, your build process is querying your OS's pkg-config for ncurses, which reports requiring linking against tinfo.  We do notbuild ncurses with an external tinfo, hence the error.

Test this patch, which should generate pkgconfig files for ncurses, and point nano at them:

```
diff --git a/release/src/router/Makefile b/release/src/router/Makefile
index eb6a627..dc8e146 100755
--- a/release/src/router/Makefile
+++ b/release/src/router/Makefile
@@ -4176,13 +4176,14 @@ ncurses/Makefile:
        $(if $(RTCONFIG_BCMARM),--enable-widec,) \
        --without-debug --without-manpages --without-profile --without-progs --without-tests \
        --disable-home-terminfo --with-normal --with-shared --with-build-cppflags=-D_GNU_SOURCE \
+       --enable-pc-files \
        --disable-termcap --disable-database --with-fallbacks="xterm,vt100,vt200,linux,ansi,xterm-256color"
 
 #      --with-default-terminfo-dir=/usr/share/terminfo --with-terminfo-dirs=/usr/share/terminfo
 
 ncurses: ncurses/Makefile
        @$(SEP)
-       cd ncurses && $(MAKE) -C ncurses
+       cd ncurses && $(MAKE) -C ncurses && $(MAKE) -C misc pc-files
 
 ncurses-clean:
        -@$(MAKE) -C ncurses distclean
@@ -4208,6 +4209,7 @@ nano/Makefile:
        cd nano && $(CONFIGURE) --prefix=$(INSTALLDIR)/nano/usr \
                CFLAGS="$(EXTRACFLAGS) -Os -I$(TOP)/ncurses/include -ffunction-sections -fdata-sections" \
                LDFLAGS="$(LDFLAGS) -L$(TOP)/ncurses/lib -ffunction-sections -fdata-sections -Wl,--gc-sections" \
+               PKG_CONFIG_LIBDIR="$(TOP)/ncurses/misc" \
                $(if $(RTCONFIG_BCMARM),ac_cv_lib_ncursesw_get_wch=yes,ac_cv_lib_ncursesw_get_wch=no) \
                $(if $(RTCONFIG_BCMARM),--enable-utf8,--enable-tiny) \
                --disable-speller --disable-extra --disable-tabcomp --enable-wrapping --disable-glibtest \

```
Don't forget to issue a "make nano-clean ncurses-clean" before testing.

commit 07603fb2 (2016-10-02): ncurses/nano: Generate ncurses pkgconfig files, and point the nano build at them - should ensure nano doesn't try to inherit the pkgconfig from the build machine (which would tell it to link against tinfo; closes #1039)
closed 2016-10-02 by RMerl