Asuswrt-Merlin/asuswrt-merlin.legacy #1043: Wifi toggle button disabling scheduling

sosandroid, 2016-09-27:
Firmware tested : 380.62 Beta
Device : RT-AC87U

When using the wifi toggle button on the front panel, it disables the wifi scheduling.

A common use case for me : The wifi is usually disabled at night. When having some homework to do urgently, I activate back the wifi before turning it off when going to bed. At this time, the sheduler is disabled.

It can be an enhancement more than a bug.
Regards

RMerl, 2016-09-27:
This is the intended behaviour - the button is intended as a global kill switch, not just a toggle.  Previously it used to work the way you asked, but a number of users complained about it.

closed 2016-09-27 by RMerl

sosandroid, 2016-09-27:
Merci pour le feedback
Is this an Idea to allow both behaviors, selected from the UI ?
Regards

RMerl, 2016-09-28:
I want to keep down on the number of configurable options, as this is one of the things people often dislike about other third party firmwares compared to Asuswrt-Merlin.  Especially something that is so specific.

sosandroid, 2016-09-28:
To be honnest, I do not understand the purpose of a kill switch here. In case of "emergency", you just have to switch off the wifi. Disabling the scheduling also seems (to me) a bit weird.

Anyway, I understand the will to keep the number of configurable options as low as possible.
Looking for a solution that is convenient for me & my wife :). __Le mouton à 5 pattes ... __