Asuswrt-Merlin/asuswrt-merlin.legacy #1045: Add Option to set the HTTP Port of the Web Interface

AndiDittrich, 2016-10-02:
It would be great if someone can add an option which allows users to set the **http** port via the GUI.
Currently, only the **https** port can be set in https only mode.

### Problem
Running other services on port 80 it is not possible, because the build-in **httpd** service is bound to **all interfaces** by default. HTTPS only mode is not a solution because it is very slow on older devices like N66U using modern browser with modern ciphers.

Another solution could be to bind the httpd service to the **lan interface** only - this allows users to add virtual interfaces for other services

RMerl, 2016-10-03:
httpd no longer binds to all interfaces, that was changed a few versions ago:

```
admin@Stargate88:/tmp/home/root# ps w | grep http
  530 admin     6508 S    httpds -s -p 8443 -i br0
  531 admin     6764 S    httpd -i br0
```

I do understand the usefulness of being able to change the port however.  I will need to see how doable this is, as the rest of the code needs to be aware of this as well (for instance, services that redirect to the httpd in case of WAN being down, or AiProtection page blocks.  It's not as simple as just specifying a port at httpd launch.

AndiDittrich, 2016-10-05:
thanks for that hint!

RMerl, 2017-05-30:
Allowing httpd's port to be changed from 80 will have to be implemented by Asus, as some of their closed-source components have it hardcoded in them.

pnwlabrat, 2018-01-03:
Dang. So, no one using this code, stock or Merlin can host a website internally on Port 80. I'm having troubles getting my Synology home server to pull a "Let's Encrypt" cert, because it reaches out over HTTP/80 back to the server. The Asus blocks access to the internal machine, listening on port 80. 

Obviously not your fault, but bummer.

RMerl, 2018-01-03:
You could just disable http support, which would free up port 80.

Anyway, Asus has made the non-ssl port configurable about a month ago, so that should be possible in the future.