Asuswrt-Merlin/asuswrt-merlin.legacy #1047: Add TOR geoip database to firmware

blackfuel, 2016-10-03:
Example patch for the Makefile to copy TOR's geoip database files into the firmware.  Then it's possible to specify the location in TOR's configuration file (torrc).  Example:
GeoIPFile /rom/tor/geoip
GeoIPv6File /rom/tor/geoip6

[asuswrt-merlin-tor-geoip.patch.txt](https://github.com/RMerl/asuswrt-merlin/files/506096/asuswrt-merlin-tor-geoip.patch.txt)

RMerl, 2016-10-03:
Those database would add nearly 800 KB each to the firmware size - can't justify adding 1.6 MB for this, sorry.

closed 2016-10-03 by RMerl

blackfuel, 2016-10-03:
Let the record show that the TOR 0.2.8.8 database files (both geoip and geoip6) increase the size of the TRX firmware image by a total 600 KB.  Apparently the SqaushFS compression is quite good.