Asuswrt-Merlin/asuswrt-merlin.legacy #1048: Authentication issue?

Diaoul, 2016-10-06:
Thanks for the great firmware alternative :)

I face a weird issue, here are the reproduce steps:

- Log in the interface
- Go to a setting page (WAN -> Virtual Server / Port Forwarding for example but any will do)
- Change something
- Press Apply
- Nothing is done and the login screen shows up

Here is what I've tried so far:

- Flash again
- Factory reset
- Reboot
- Disconnect all other devices
- Erase browser cache and cookies

I don't have this issue on the official firmware. The weird part is that sometimes it works, sometimes it doesn't and I can't seem to find a pattern...

RMerl, 2016-10-16:
Try accessing your router's interface through its IP, and make sure your security software isn't interfering with the router (some, like Eset Internet Security do).

closed 2016-10-16 by RMerl

Diaoul, 2016-10-16:
There is one way to be able to access it that works:

* Plug my computer directly on the router instead of through a switch
* Reboot the router
* Use the IP instead of a DNS entry

What is strange is that it works without any issue on the stock firmware. Did you change anything related to login operations? Where is the code for the authentication part?