Asuswrt-Merlin/asuswrt-merlin.legacy #1049: Minor adjustments for README.TXT

nielspeen, 2016-10-07:
While re-creating the build environment by following the steps in README.TXT, I ran into a few minor issues:

- When building RT-N66U, the build process wanted automake 1.10 instead of automake 1.15 (1.15 is default for Ubuntu 16.04.) Installing the automake1.11 package fixed this (1.11 is also available in Ubuntu 16.04.)
- When building RT-56U I needed to install libltdl-dev or netatalk would fail to build.
- For reasons I do not fully understand, I had to perform the steps outlined in post #6 of this thread: http://www.snbforums.com/threads/compiling-the-firmware-for-rt-ac56u.24040/#post-180456 .  
- To build RT-AC3200 I had to replace "make rt-ac3200u" with "make rt-ac3200".

ecrsoftware, 2016-10-17:
In regards to adjustments for the README, I too ran into issues with the stock guide, but using the suggested Linux Mint 13 (per https://asuswrt.lostrealm.ca/source). In order to get my build environment working properly, on 380.63_alpha1, I had to install the following packages after performing all tasks outlined in the guide:
```
sudo apt-get install automake
sudo apt-get install autopoint
sudo apt-get install gettext
sudo apt-get install libgtk2.0-dev
```
I then had to reconfigure nano, as its stock was hard-coded to automake 1.15 instead of 1.11 that comes default from apt sources in Linux Mint 13/Cinnamon
```
cd /home/asus/asuswrt-merlin/release/src/router/nano
autoreconf --force --install
```

You can view the errors I received and researched in order to determine what packages were missing and that nano needed to be reconfigured with the automake version on my system at the following gist: https://gist.github.com/ecrsoftware/db11a6a5e624565b6dc6ba0254a00eca

RMerl, 2016-10-17:
I moved the build environment to Ubuntu 16.04 LTS this summer.  While I updated most documentation, it`s possible that some of it (like the website) have slipped through the tracks.

The Wiki will always be the definitive place to look for up-to-date documentation there.

Commit 9329fa6c72c335097bca26aa6e76c2f478b87fdf should in theory take care of the problem for people still using older distros.

closed 2016-11-25 by RMerl