Asuswrt-Merlin/asuswrt-merlin.legacy #1050: Feature request : Disable port forwards option

neoatomic, 2016-10-07:
@RMerl First thanks for all the time that you put into this project. 

Every few weeks i need to open/forward a port for an incoming FTP server. Currently i need to re-add/remove the port with every use. (as i want the port to be closed after usage)
I'm hoping it would be possible to have an option to just "disable" a individual forwarded port, so that the settings are saved, and you can reactivate it with a simple mouse click.

p3yot3, 2016-10-07:
I would also love to see this feature implemented for similar reasons.

Thanks & great work!

fanaticDavid, 2016-10-08:
I would also love this feature. My previous router, a CISCO Linksys E4200, had this in the standard firmware. Really convenient!

HGWJ, 2016-10-12:
I would also love to see this feature implemented for similar reasons.

RMerl, 2016-10-14:
Unlikely, as this would break compatibility with the stock firmware.  This is something that would have to be implemented by Asus ideally.

closed 2016-10-14 by RMerl