Asuswrt-Merlin/asuswrt-merlin.legacy #1051: Subnet change causes web UI address change ... service becomes unavailable

mypluribus, 2016-10-11:
Hello, not sure if this is a bug or user error ...

When setting up the router (RT-AC66U, after reset) w/ asus-merlin 380.62_1, I found that if I change the LAN subnet and enable HTTPS, the web interface becomes unavailable at both the new address and the old address ... router.asus.com:8443 also does not work.

I can't remember the message that is displayed (and I can't find a screenshot online) but it states something to the effect of, "the IP address has been changed and the website will reload" ... "your connection has been terminated".

The process I used was (directly after reset ... skip the setup wizard):
1.) login to the web interface 
2.) go to the LAN tab 
3.) change the IP address of the device from 192.168.1.1 to 10.10.10.1
4.) click apply and wait (at this point the web service is now available @ 10.10.10.1)
5.) click Administration --> System and select HTTPS on the Authentication method drop down. 

This is when the web service becomes unavailable @ 10.10.10.1.

Everything works as expected if I don't change the LAN subnet ... however, this won't work for me since this device needs to join a network that already has this subnet in use.

Thanks in advance for any help you can give me.

RMerl, 2016-10-16:
Works for me, as I'm using a different LAN subnet with https enabled.  Try accessing through the router's IP rather than router.asus.com, your computer's resolver probably has the old IP cached.

closed 2016-10-16 by RMerl

mypluribus, 2016-10-16:
Hello and thank you for your response.

I have edited the original issue to be more clear that I am accessing the device using its IP address and only used router.asus.com as another method to access the router web interface to verify it's operation.  So, this is not a resolver caching issue, since I'm only resolving the hostname as a last resort and using an IP address as the primary method of access.

Regardless, this does not work for my Asus RT-AC66U.  I have tried multiple times.  I'll try to dig into the issue when I have time.  In the mean time, I guess I'll revert back to the Asus firmware.  Too bad, I had high hopes.

Best of luck with your project.