Asuswrt-Merlin/asuswrt-merlin.legacy #1053: No IPv6 for AP Mode

chrisstaite, 2016-10-12:
I have no option for IPv6 configuration of the router when it is configured in AP Mode.

I can ping using IPv4 from the SSH but get ping6: sendto: Network is unreachable when trying IPv6.  I imagine that IPv6 traffic going over the AP to my router is being dropped as it does not know the route.

I am having issues with my devices (Android Nougat) saying no access to the Internet when it is connected to the AP.  I believe this may be the cause.

charlie2alpha, 2016-10-26:
That is not a bug. In AP mode no routing or firewalling is done, if your actual router supports IPv6 the clients should pick it up automatically. If not, look at your router and your devices, the AP is a simple bridge and has nothing to do with it.

chrisstaite, 2016-10-26:
Something isn't working because IPv6 wouldn't work when clients were associated with the AP running this firmware, but will work when the device is running dd-wrt.  I imagine it isn't passing the IPv6 through for some reason.  IPv6 is running natively on the OpenWRT router, OpenWRT AP and this router when running dd-wrt (OpenWRT not available for it).

amosbird, 2016-11-15:
I have a similar same problem. My router in AP mode cannot get an ipv6 addr while my client does. An v6 addr is needed for my router.

RMerl, 2016-11-25:
I don't touch IPv6 anymore.  Any issue will have to be resolved upstream - I have no way of testing this properly.

closed 2016-11-25 by RMerl