Asuswrt-Merlin/asuswrt-merlin.legacy #1054: Policy based VPN routing does not seem to work as expected RT-N66U 380.62_1

ashafer01, 2016-10-13:
With source and destination IP both set to "0.0.0.0" and Iface set to VPN, I'm not seeing any traffic get routed over the VPN connection, both if this is the only rule and if there are others. Everything works as expected by setting "Redirect Internet Traffic" to "All".

My goal is to add an exception to have a particular subnet routed over the WAN and everything else routed over the VPN by default.

moritzfl, 2016-10-13:
Have you tried to set the source in cidr format? That would allow you to cover an ip-range.

ashafer01, 2016-10-14:
Using 0.0.0.0/0 still does not seem to work as expected for a source, however using my LAN subnet works and is sufficient for my purposes.

closed 2016-10-14 by ashafer01