Asuswrt-Merlin/asuswrt-merlin.legacy #1055: Bring back support for RT-N16

hubertbanas, 2016-10-13:
I see Asus pushed out 9.0.0.4.380.2985 version of their firmware. Is there any chance to bring back support for RT-N16?

RMerl, 2016-10-14:
None, for many reasons.  First one is you need closed source components from a compatible firmware.  The 2985 components for instance are not compatible with the GPL 3xxx code currently used by Asuswrt.

Also, the RT-N16 is an old model, without sufficient nvram (32 KB) to support modern features.  It's also using an older Broadcom SDK that requires to be developped specifically for it.

closed 2016-10-14 by RMerl

hubertbanas, 2016-10-14:
Fair enough. Thanks for the explanation.

cub-uanic, 2016-10-23:
@RMerl do you have any recommendations how to try to build fw based on your work for N16?
I'd like to get initial directions from you and to try to build it myself.
Any help or advise is really appreciated.