Asuswrt-Merlin/asuswrt-merlin.legacy #1057: Bugged WAN > Internet Connection config

apazga, 2016-10-15:
Hi,

I think I found a bug that appears when you hit "Apply" in WAN > Internet Connection. If I change any value/option before hitting Apply, changes are not saved and the following is logged. If I didn't change any value, and just hit Apply, same happens.

(**Asus RT-AC87U using Asuswrt-Merlin: 380.62_1**)
**Log**
```
Oct 15 01:13:11 rc_service: httpd 477:notify_rc restart_wan_if 0
Oct 15 01:13:12 pppd[526]: Hangup (SIGHUP)
Oct 15 01:13:12 pppd[526]: Terminating on signal 15
Oct 15 01:13:12 pppd[526]: Connect time 2.2 minutes.
Oct 15 01:13:12 dnsmasq[471]: read /etc/hosts - 5 addresses
Oct 15 01:13:12 pppd[526]: Sent 885659 bytes, received 1287843 bytes.
Oct 15 01:13:12 dnsmasq[471]: read /etc/hosts.dnsmasq - 28 addresses
Oct 15 01:13:12 dnsmasq-dhcp[471]: read /etc/ethers - 28 addresses
Oct 15 01:13:12 miniupnpd[832]: Failed to get IP for interface ppp0
Oct 15 01:13:12 miniupnpd[832]: SendNATPMPPublicAddressChangeNotification: cannot get public IP address, stopping
Oct 15 01:13:12 dnsmasq[471]: using nameserver 80.58.61.250#53
Oct 15 01:13:12 dnsmasq[471]: using nameserver 80.58.61.254#53
Oct 15 01:13:12 pppd[526]: Connection terminated.
Oct 15 01:13:12 pppd[526]: Sent PADT
Oct 15 01:13:12 dnsmasq[471]: read /etc/hosts - 5 addresses
Oct 15 01:13:12 dnsmasq[471]: read /etc/hosts.dnsmasq - 28 addresses
Oct 15 01:13:12 dnsmasq-dhcp[471]: read /etc/ethers - 28 addresses
Oct 15 01:13:12 dnsmasq[471]: using nameserver 80.58.61.250#53
Oct 15 01:13:12 dnsmasq[471]: using nameserver 80.58.61.254#53
Oct 15 01:13:12 kernel: Interface ppp0 doesn't exist
Oct 15 01:13:12 pppd[526]: Exit.
Oct 15 01:13:15 wan: mac clone: [wan0_hwaddr] == [00:AA:AA:AA:AA:AA]
Oct 15 01:13:15 pppd[915]: Plugin rp-pppoe.so loaded.
Oct 15 01:13:15 pppd[915]: RP-PPPoE plugin version 3.11 compiled against pppd 2.4.7
Oct 15 01:13:15 pppd[917]: pppd 2.4.7 started by apazga, uid 0
Oct 15 01:13:17 pppd[917]: PPP session is 777 (0x309)
Oct 15 01:13:17 pppd[917]: Connected to 00:BB:BB:BB:BB:BB via interface vlan6
Oct 15 01:13:17 pppd[917]: Using interface ppp0
Oct 15 01:13:17 pppd[917]: Connect: ppp0 <--> vlan6
```
(Just replaced MAC Addresses)

**Screenshot**
[image]

If I can do any other test, please just tell me.
Hope it helps and thank you for your work.

Best regards!
Abel.

RMerl, 2016-10-16:
Works for me.  Try with a different browser.

closed 2016-10-16 by RMerl

apazga, 2016-10-16:
Previous one was using Chrome
This one is using Firefox:

```
Oct 16 12:27:00 rc_service: httpd 477:notify_rc restart_wan_if 0
Oct 16 12:27:00 dnsmasq[471]: read /etc/hosts - 5 addresses
Oct 16 12:27:00 dnsmasq[471]: read /etc/hosts.dnsmasq - 28 addresses
Oct 16 12:27:00 dnsmasq-dhcp[471]: read /etc/ethers - 28 addresses
Oct 16 12:27:00 dnsmasq[471]: using nameserver 80.58.61.250#53
Oct 16 12:27:00 dnsmasq[471]: using nameserver 80.58.61.254#53
Oct 16 12:27:00 pppd[917]: Hangup (SIGHUP)
Oct 16 12:27:00 pppd[917]: Terminating on signal 15
Oct 16 12:27:00 pppd[917]: Connect time 2113.7 minutes.
Oct 16 12:27:00 pppd[917]: Sent 383145356 bytes, received 1977015243 bytes.
Oct 16 12:27:00 miniupnpd[963]: Failed to get IP for interface ppp0
Oct 16 12:27:00 miniupnpd[963]: SendNATPMPPublicAddressChangeNotification: cannot get public IP address, stopping
Oct 16 12:27:00 pppd[917]: Connection terminated.
Oct 16 12:27:00 pppd[917]: Sent PADT
Oct 16 12:27:01 dnsmasq[471]: read /etc/hosts - 5 addresses
Oct 16 12:27:01 dnsmasq[471]: read /etc/hosts.dnsmasq - 28 addresses
Oct 16 12:27:01 dnsmasq-dhcp[471]: read /etc/ethers - 28 addresses
Oct 16 12:27:01 dnsmasq[471]: using nameserver 80.58.61.250#53
Oct 16 12:27:01 dnsmasq[471]: using nameserver 80.58.61.254#53
Oct 16 12:27:01 kernel: Interface ppp0 doesn't exist
Oct 16 12:27:01 pppd[917]: Exit.
Oct 16 12:27:03 wan: mac clone: [wan0_hwaddr] == [00:AA:AA:AA:AA:AA]
Oct 16 12:27:03 pppd[6285]: Plugin rp-pppoe.so loaded.
Oct 16 12:27:03 pppd[6285]: RP-PPPoE plugin version 3.11 compiled against pppd 2.4.7
Oct 16 12:27:03 pppd[6287]: pppd 2.4.7 started by adminuser, uid 0
Oct 16 12:27:03 pppd[6287]: PPP session is 5893 (0x1705)
Oct 16 12:27:03 pppd[6287]: Connected to 00:BB:BB:BB:BB:BB via interface vlan6
Oct 16 12:27:03 pppd[6287]: Using interface ppp0
Oct 16 12:27:03 pppd[6287]: Connect: ppp0 <--> vlan6
Oct 16 12:27:06 pppd[6287]: PAP authentication succeeded
Oct 16 12:27:06 pppd[6287]: peer from calling number 00:30:88:19:AE:5C authorized
Oct 16 12:27:06 miniupnpd[963]: Failed to get IP for interface ppp0
Oct 16 12:27:06 miniupnpd[963]: SendNATPMPPublicAddressChangeNotification: cannot get public IP address, stopping
Oct 16 12:27:06 miniupnpd[963]: Failed to get IP for interface ppp0
Oct 16 12:27:06 miniupnpd[963]: SendNATPMPPublicAddressChangeNotification: cannot get public IP address, stopping
Oct 16 12:27:06 miniupnpd[963]: Failed to get IP for interface ppp0
Oct 16 12:27:06 miniupnpd[963]: SendNATPMPPublicAddressChangeNotification: cannot get public IP address, stopping
Oct 16 12:27:06 pppd[6287]: local  IP address XX.XX.XX.XX
Oct 16 12:27:06 pppd[6287]: remote IP address YY.YY.YY.YY
Oct 16 12:27:06 pppd[6287]: primary   DNS address 80.58.61.250
Oct 16 12:27:06 pppd[6287]: secondary DNS address 80.58.61.254
Oct 16 12:27:06 miniupnpd[963]: ioctl(s, SIOCGIFADDR, ...): Cannot assign requested address
Oct 16 12:27:06 miniupnpd[963]: Failed to get IP for interface ppp0
Oct 16 12:27:06 miniupnpd[963]: SendNATPMPPublicAddressChangeNotification: cannot get public IP address, stopping
Oct 16 12:27:06 rc_service: ip-up 6295:notify_rc start_firewall
Oct 16 12:27:06 dnsmasq[471]: read /etc/hosts - 5 addresses
Oct 16 12:27:06 dnsmasq[471]: read /etc/hosts.dnsmasq - 28 addresses
Oct 16 12:27:06 dnsmasq-dhcp[471]: read /etc/ethers - 28 addresses
Oct 16 12:27:06 dnsmasq[471]: using nameserver 80.58.61.250#53
Oct 16 12:27:06 dnsmasq[471]: using nameserver 80.58.61.254#53
Oct 16 12:27:07 start_nat_rules: apply the nat_rules(/tmp/nat_rules_0_ppp0_vlan6)!
Oct 16 12:27:07 wan: finish adding multi routes
Oct 16 12:27:07 rc_service: ip-up 6295:notify_rc stop_upnp
Oct 16 12:27:07 rc_service: waitting "start_firewall" via ip-up ...
Oct 16 12:27:08 rc_service: ip-up 6295:notify_rc start_upnp
Oct 16 12:27:08 rc_service: waitting "stop_upnp" via ip-up ...
Oct 16 12:27:08 miniupnpd[963]: shutting down MiniUPnPd
```

gilstrac, 2016-10-17:
It hasn't worked in chrome for me for a few versions now but it works in
firefox.