Asuswrt-Merlin/asuswrt-merlin.legacy #1058: https port 443

damccull, 2016-10-17:
When trying to change the port for https webui to 443, the router informs me the port is reserved for 'cloud disk'. I do not use cloud disk and I would rather put my https interface on 443. Can you add a way to do that in a future version please?

JamieMagee, 2016-10-17:
It's possible currently. You just have to change the Cloud Disk port (Even if Cloud Disk is disabled). After that you can change the HTTPS port for the webui to 443.

RMerl, 2016-12-07:
As mentionned, not a limitation, just a configuration issue.  THe router ensures that you use unique ports to all services, so they do not conflict with one another.  Since AiCloud defaults to 443, you have to move that service to a different port first to free it.

closed 2016-12-07 by RMerl