Asuswrt-Merlin/asuswrt-merlin.legacy #1060: Improve default DropBear config

JamieMagee, 2016-10-17:
A quick scan of the default DropBear config using [ssh-audit](https://github.com/arthepsy/ssh-audit) gives the following suggestions:

```
# algorithm recommendations (for Dropbear SSH 2016.74)
(rec) -ecdh-sha2-nistp521           -- kex algorithm to remove
(rec) -ecdh-sha2-nistp384           -- kex algorithm to remove
(rec) -ecdh-sha2-nistp256           -- kex algorithm to remove
(rec) -diffie-hellman-group1-sha1   -- kex algorithm to remove
(rec) -diffie-hellman-group14-sha1  -- kex algorithm to remove
(rec) +diffie-hellman-group16-sha512-- kex algorithm to append
(rec) +diffie-hellman-group14-sha256-- kex algorithm to append
(rec) -ecdsa-sha2-nistp521          -- key algorithm to remove
(rec) -ssh-dss                      -- key algorithm to remove
(rec) -twofish128-cbc               -- enc algorithm to remove
(rec) -twofish256-cbc               -- enc algorithm to remove
(rec) -3des-cbc                     -- enc algorithm to remove
(rec) -aes256-cbc                   -- enc algorithm to remove
(rec) -twofish-cbc                  -- enc algorithm to remove
(rec) -aes128-cbc                   -- enc algorithm to remove
(rec) +twofish256-ctr               -- enc algorithm to append
(rec) +twofish128-ctr               -- enc algorithm to append
(rec) -hmac-sha1                    -- mac algorithm to remove
(rec) -hmac-sha1-96                 -- mac algorithm to remove
(rec) -hmac-md5                     -- mac algorithm to remove
```

The full output is [here](https://gist.github.com/JamieMagee/d9231818fd2aff91eef60b3ca56276d7)

JamieMagee, 2016-10-17:
Yeah, I ran this against my own RT-AC3200 running ASUSWRT-Merlin 380.62-1.

RMerl, 2016-10-17:
I'll have to work through the list to decide what to drop.  Some of these changes I disagree with, as they seem to expect everyone to be running either a recent OpenSSH or dropbear client, with an up-to-date OpenSSL version.  Some mobile devices might be using older versions.  And there are a lot of devices out there still running OpenSSL 0.9.8.  Also, this audit didn't test non-OpenSSL clients such as XShell or SecureCRT.  So, I'd rather be conservative with supported protocols.

For instance, I see nothing wrong with supporting AES-128-CBC or AES-256-CBC from a security point of view - those ciphers are still considered secure.

Things like hmac-md5 however should be removed.

JamieMagee, 2016-10-17:
Yeah, I did a quick bit of research and found that AES-128-CBC and AES-256-CBC were dropped from OpenSSH in version 6.7. However, Ubuntu 14.04 LTS is still on OpenSSH 6.6, and is supported until 2019.

RMerl, 2016-10-17:
As an added reference, here's a list of what is supported by Xshell5 (the client that I use):

[image]
[image]
[image]

RMerl, 2016-10-21:
I checked the dropbear config, and compared everything against RFC4253 which defines the SSH protocol.  Quite a few of these deprecated ciphers/macs are marked as mandatory by the protocol unfortunately, like SHA1 support.

I'll remove MD5 since it's not mandatory.

Twofish-CTR is marked as "untested" and disabled by default - the dropbear author is looking for feedback to determine if it's working properly, so I'll keep it disabled.

Breaking RFC specifications might be fine for a server admin disabling unsafe protocols, but for an out-of-the-box configuration, I prefer to stick as much as possible to RFC requirements.

RMerl, 2016-10-24:
I am reconsidering this.  Gathering user feedback, and so far I tend to side with them - security is more important here, even if ultimately an end-user might simply configure his client to use a more secure hmac first.  Therefore I will possibly also remove other weak points such as hmac-sha1 and dh-group1, pending some review first.  I will probably use RHEL5 as a baseline for the oldest supported SSH client.

RMerl, 2016-10-27:
According to various studies, hmac-sha1 is not sensitive to collisions due to the way hmac works.

JamieMagee, 2016-10-27:
Yeah, I have read the same things. I think `ssh-audit` suggests disabling it as a precautionary measure, rather than because of any actual attack on `hmac-sha1`

RMerl, 2016-11-14:
After reviewing a lot of documentation and polling feedback from people more knowledgeable in ciphers, I've decided not to make any further change at this time.  hmac-sha1 is still secure due to the private part of the source material, and RFCs still mention that it's a mandatory element of the SSH protocol.

It was worth the review however, just to be sure, so thanks for starting the discussion.

closed 2016-11-14 by RMerl