Asuswrt-Merlin/asuswrt-merlin.legacy #1061: /jffs/configs/group.add does not work

cbz, 2016-10-17:
Hi - I have the following contents in /jffs/configs/group.add:

nogroup:x:5:

After a reboot, this isn't added to the contents of /tmp/etc/group|/etc/group.  I do have "Enable JFFS custom scripts and configs" set to 'Yes', and I am already using custom configs for dnsmasq.conf (which works).

The reason I need this btw is because the avahi daemon in opkg tries to chgrp to 'nogroup' after it is started up - and this fails.  I notice there is an avahi daemon in the github source tree - not sure if this is now built into the official build, I can't find it at any rate.

I tried posting this to the snbforums instead, but when I do cloudflare blocks me (apparently my post may contain sql commands).

cbz, 2016-10-18:
Incidentally, this is with version 380.62_1 of the firmware - though it doesn't work with older firmware either.

RMerl, 2016-10-20:
Issue occurs because the passwd/group/shadow files are configured before jffs2 is mounted.

closed 2016-10-20 by RMerl

cbz, 2016-10-20:
but if you are recreating this - then I should see the change applied afterwards, shouldn't I? In this case even after a successful boot the .add file isn't added to /etc/group

RMerl, 2016-10-20:
I fixed it last night, so unless you just recompiled the firmware from the Git repo, you won't see this change until the next release.

commit 74eeda2e (2016-10-20): rc: re-create passwd/shadow/group after the JFFS2 partition is mounted, in case we have custom configs stored there (fixes #1061)