Asuswrt-Merlin/asuswrt-merlin.legacy #1063: Building ipset_arm tries to link to wrong libmnl.so

charlie2alpha, 2016-10-19:
In my system and build environment when building the ipset_arm component of the firmware instead of linking with the locally built libmnl.so it tries to link to my system's native lib which of course won't work and aborts the process with the linker complaining about incompatible binary format.

I find a solution for this, which I propose with this patch: http://pastebin.com/8siWia0i

Works perfectly fine for my build environment.

RMerl, 2016-10-20:
I've never seen such a syntax used with -l, is that actually valid?

charlie2alpha, 2016-10-20:
From ld's man page:

-l namespec
--library=namespec
Add the archive or object file specified by namespec to the list of files to link. This option may be used any number of times. If namespec is of the form :filename, ld will search the library path for a file called filename, otherwise it will search the library path for a file called libnamespec.a.

RMerl, 2016-10-20:
The manpage I'm currently looking at from the GNU website has no mention of the ":filename" syntax.

```
-Lsearchdir
--library-path=searchdir
Add path searchdir to the list of paths that ld will search for archive libraries and ld control scripts. You may use this option any number of times. The directories are searched in the order in which they are specified on the command line. Directories specified on the command line are searched before the default directories. All -L options apply to all -l options, regardless of the order in which the options appear. The default set of paths searched (without being specified with `-L') depends on which emulation mode ld is using, and in some cases also on how it was configured. See section Environment Variables. The paths can also be specified in a link script with the SEARCH_DIR command. Directories specified this way are searched at the point in which the linker script appears in the command line.
```

charlie2alpha, 2016-10-20:
well, the man page in my own system has it, and the first hit for "man ld" in Google returns https://linux.die.net/man/1/ld which does have the syntax. Also:

# ld -v
GNU ld (Gentoo 2.25.1 p1.1) 2.25.1

This my system's linker version.

charlie2alpha, 2016-10-20:
If you prefer not to risk it there's an alternate approach to this, since the .la file causes the problem, use this patch to get rid of it, the linker will fallback to the .so file automatically. Simple patch here: http://pastebin.com/tZWdSUXD

RMerl, 2016-10-21:
The version that matters is the toolchain's, not your system's:

```
arm-brcm-linux-uclibcgnueabi-ld -V
GNU ld (GNU Binutils) 2.21.1
  Supported emulations:
   armelf_linux_eabi
   armelfb_linux_eabi
```

I suspect the manpage I consulted was for an older version that didn't implement this syntax yet.  The BCM toolchain's LD does supports it, so I'm fine with using that syntax.  I don't understand however why your build environment would try to link with the wrong version - it's working fine for three different firmware projects so far.

commit 343d6f97 (2016-10-21): ipset_arm: some build systems try to link against the wrong library, ensure we do link against our built libmnl.so (closes #1063)
closed 2016-10-21 by RMerl