Asuswrt-Merlin/asuswrt-merlin.legacy #1066: Dirty COW fixed needed?

blackfuel, 2016-10-21:
I could not find a Dirty COW patch for Linux 2.6.36.4 on kernel.org, so I made this one myself.  My RT-AC68U is running fine with this patch applied and it does not crash.  

[asuswrt-merlin-ARM-DirtyCOW.patch.txt](https://github.com/RMerl/asuswrt-merlin/files/544154/asuswrt-merlin-ARM-DirtyCOW.patch.txt)

RMerl, 2016-10-21:
Thanks.  I'll extend the patch to also cover the RT-AC3200 kernel (under src-rt-7.x.main).

yasij, 2016-10-22:
This also affects the 2.6.22 kernels: https://github.com/dirtycow/dirtycow.github.io/wiki/VulnerabilityDetails

I'm working on a backport from the 3.2.83 kernel version:
[0001-kernel-mips-Patch-the-2.6.22.19-kernel-against-CVE-2.patch.txt](https://github.com/RMerl/asuswrt-merlin/files/545936/0001-kernel-mips-Patch-the-2.6.22.19-kernel-against-CVE-2.patch.txt)

I am currently building an RT-AC66U to test it.

RMerl, 2016-10-23:
At first glance the patch looks good to me.  How did your tests go?

yasij, 2016-10-23:
I've been having stability issues on my Linux machine running 4.8.4 when trying to build the image for RT-AC66U. Loads that ends up mmaping a bunch of files like that build or "lsof +D /home" eventually results in "BUG: Bad page map in process <x>" with a bunch of stuck processes. I'm not sure if this is a side effect of the security fix or something else. I just managed to get an image built to test which I will try shortly.

RMerl, 2016-10-24:
I committed the MIPS patch (f9aa712b43c36a7817f92d7e3cf31ca5a9ad077c) - best way to gather feedback I think.

RMerl, 2016-10-25:
MIPS patch seem to be fine so far.  I don't have a test case however to confirm that it does resolve the issue, but the patch is similar enough to upstream that I'd be confident that it does.

commit 7a95789b (2016-10-21): kernel-arm: Patched all three 2.6.36 kernels against CVE-2016-5195 (Closes #1066)
closed 2016-10-25 by RMerl