Asuswrt-Merlin/asuswrt-merlin.legacy #1068: Does the current source tree support ASUS RT-AC1200G+?

clockzhong, 2016-10-22:
I saw your latest source codes already has some places to mention this device, e.g. in the 
asuswrt-merlin/release/src/router/shared/model.c
.....
{ "RT-AC1200G+", MODEL_RTAC1200GP},
......

But I didn't see the RT-AC1200G+ is in your supported devices list, and you help me confirm whether your current release could support this type device? I could help your test and debug features for your team on this device if it could be built now.

Thanks in advance!

Clock

RMerl, 2016-10-22:
It doesn't, and there are no plans to support it.

closed 2016-10-22 by RMerl

clockzhong, 2016-10-22:
RMerl,

Thanks for your reply so quickly! Why you have no plan to do it? Any difficulties you've encountered when you try it? If you haven't tried it yet, I want to try it.

Thanks!

Clock

RMerl, 2016-10-22:
To be supported, a router needs to meet a few criteria:

1) It must be based on Broadcom
2) It must be a high-end model (this router only has 16 MB of flash and 128 MB RAM)
3) It must be available worldwide, not just in a few select regions
4) I must own one
5) Must be a model for which Asus regularly updates the closed-source components required to compile a third party firmware, as I'm totally dependent on them for this, unlike Tomato or DD-WRT
6) Since this is a one-man project doing everything, I have to limit the total number of supported models to be able to fit it within my spare time.  The RT-N16 had to be dropped in part because of this.

The goal of this project is to support a small, limited number of high-end models, but to have them 100% working.

clockzhong, 2016-10-22:
RMerl,

I really appreciate your detailed explanation. I've another question if you have time to answer: Where you get the AsusWrt latest source tree? I mean full source tree? I could find them through asus official site:
https://www.asus.com/support/Download/11/2/0/152/r4fqhXAPrMxenYoF/8/
and my RT-AC1200G+ source tree could be downloaded through http://dlcdnet.asus.com/pub/ASUS/wireless/RT-AC1200G+/GPL_RT_AC1200G_PLUS_30043801234.zip

But after I extracted this zip file, and build the source tree by following the instructions in its README. There are many mistakes and mismatches between its document and the source tree. 
And this source codes package is only for one model of the asus router.
Can you tell me another place to download the full source tree of official AsusWRT? 

Thanks in advance!

Clock

RMerl, 2016-10-22:
There's no full source tree available to the public.  I have to merge code from each GPL releases for different models (hence the fact that multiple model support is a lot of work), closed source components from every single model's GPL release.  Each GPL release is downloaded from their support site.

GPL releases often have missing parts.  I have to either reuse parts from previous releases, or contact Asus directly for the missing bits if they are absolutely necessary.

clockzhong, 2016-10-22:
Oh! You did a so great work!!! and a painful work :+1: 

I asked the same question in 
http://www.snbforums.com/forums/asuswrt-official.51/
I found you also visit there frequently. You could contact with Asus through that forum?