Asuswrt-Merlin/asuswrt-merlin.legacy #1069: Just for Information - MerlinWRT on DSL-AC68U

chimcen, 2016-10-26:
Hi,

thanks for your har work an this cool stuff here :)
Just for information - i've tried to flash the latest MerlinWRT (380.62_1) to my Asus DSL-AC68U - all said it might brick the router - it works perfectly.

One thing is the DSL-modem part - its there in the menues - but i didn't have tried it, because i'm  not using a DSL connection. Could be that this part doesn't work...

Cheers

RMerl, 2016-10-26:
The DSL components won't work, and there might be other issues with hardware-specific things that could be different, like LED GPIOs.

closed 2016-10-26 by RMerl

chimcen, 2016-10-27:
Ok... could there also be a problem with wired connections, too?
I've noticed every 24h i've get a stuck connection to my wired connected cable-modem (used at dual wan on LAN1). After i restart my modem (only the modem) all will be ok for the nex 24h...