Asuswrt-Merlin/asuswrt-merlin.legacy #1070: Password are stored in cleartext

Zulgrib, 2016-10-26:
Storing passwords in cleartext is not good practice, they should be hashed.

Passwords appears in cleartext in both acc_list and http_passwd

RMerl, 2016-10-26:
Changing this would break backward compatibility, and introduce issues such as locking you out of your router if changing back to Asus's stock firmware or an older firmware.  Therefore it's not something I can easily change, it will have to be done upstream by Asus.

closed 2016-10-26 by RMerl

Zulgrib, 2016-10-26:
Can't the switching back to ASUS firmware be resolved by reset in this case ?

RMerl, 2016-10-26:
Switching back to either Asus's stock firmware or an older version of my firmware will require doing a factory default reset, wiping out all your configuration.  That's something I want to avoid, since one of the advantages of this firmware is the ease of moving back and forth between versions, compared to a pure third party such as DD-WRT or Tomato which makes a factory default reset mandatory.

Zulgrib, 2016-10-27:
Okay, contacted the people in charge of the routers at ASUS France, but they are not a security team, if you have a better contact entry for that matters I would gladly contact them.