Asuswrt-Merlin/asuswrt-merlin.legacy #1072: WLAN MAC filtering is ignored in repeater mode

Zulgrib, 2016-10-26:
When in repeater mode, the router ignores the WLAN MAC filter and the MAC filter tab is hidden.
The router doesn't use parent's AP MAC filter.

The router in repeater mode should either : 
- Have it's on MAC filter list working
- Use parent's AP MAC filter

RMerl, 2016-10-28:
The AP has no control over the configuration of the parent router - an AP is nothing more than just a network bridge, so it has no way of knowing whatever configuration exists on the upstream router.  If Asus removes the filter list page it's probably because it doesn't work while in AP mode.  Since that list is handled internally by the closed source driver, there's nothing I can do about it.

closed 2016-10-28 by RMerl

Zulgrib, 2016-11-13:
From what I understand I need to flash ASUS stock firmware to have MAC filtering in repeater mode.

> As to the missing wireless mac filter on the GUI, as I have understand from your email, you flashed the firmware with an asus wrtmerlin firmware. try flashing the repeater with the standard firmwares from the website of RT AC87U. But after that you will need to reset the repeater and reconfigure back all the settings again.
> 
> Thank you for choosing ASUS.