Asuswrt-Merlin/asuswrt-merlin.legacy #1073: Web server does not respond when using IP restriction

Zulgrib, 2016-10-26:
When using router's IP authorized IP list and connecting to the web gui over https from a non authorized IP, the web gui asks for credentials, and once entered, the web server does not respond to anyone until rebooted from "reboot" in ssh, or from switching off and on with power switch.

I was not able to collect associated logs from web gui during this situation ...
I can try to collect them via SSH if required logs path are shared, and if #1071 is resolved

RMerl, 2016-10-28:
Can't reproduce here.  When an IP isn't authorized, I don't even reach the login page, Chrome immediately (after accepting the SSL certificate) returns:

```
The 192.168.1.1 page isn’t working

192.168.1.1 didn’t send any data.
ERR_EMPTY_RESPONSE
```

What browser are you using?

RMerl, 2016-10-28:
Also, try a test build from https://www.mediafire.com/folder/bj94sbhrh7e49/Test_Builds , in case your issue might be resolved by commit 97ba45269f1f612dd0cebd1a09f01970f0c9f026 .

closed 2016-11-06 by RMerl

Zulgrib, 2016-11-07:
Was not able to reproduce with test build