Asuswrt-Merlin/asuswrt-merlin.legacy #1077: Force poke ISP on boot to maintain IPv6 prefix (via dhcp-pd)

nimish, 2016-10-27:
Sky UK requires that you make an active prefix request, so even if you disable sending RELEASE on exit, it won't work on the next boot. 

See: http://www.snbforums.com/threads/ipv6-duid.34523/page-2

RMerl, 2016-10-28:
Disabling the RELEASE sending was just the matter of adding an option to control an odhcp6c existing setting.  This other issue however requires working around a quirk from the ISP, which I have no way of testing (and don't really have the patience to deal with either).  Someone else will have to address this.

closed 2016-11-25 by RMerl

nimish, 2016-11-27:
I took a look internally and it is indeed quite complex. Not really a quick fix at all without modifying odhcp6c or using something else. 

WORKAROUND: kill odhcp6c at start and restart it using a script