Asuswrt-Merlin/asuswrt-merlin.legacy #1079: CD-ROM supporting?

Nic2012, 2016-10-30:
Hi
when I add a CD-ROM drive to USB-Hub it seems the Asus ac56u router is recognized it but wont mount it:

Oct 30 18:29:34 kernel: usb 2-2.4.3: new high speed USB device using ehci_hcd and address 11
Oct 30 18:29:34 kernel: scsi13 : usb-storage 2-2.4.3:1.0
Oct 30 18:29:35 kernel: scsi 13:0:0:0: CD-ROM            HL-DT-ST BD-RE  BH10LS30  1.00 PQ: 0 ANSI: 0
Oct 30 18:29:35 kernel: sr0: scsi3-mmc drive: 40x/40x writer dvd-ram cd/rw xa/form2 cdda tray
Oct 30 18:29:35 kernel: cdrom: Uniform CD-ROM driver Revision: 3.20
Oct 30 18:29:36 kernel: sr 13:0:0:0: Attached scsi generic sg6 type 5
Oct 30 18:29:44 kernel: cdrom: Uniform CD-ROM driver unloaded

Any chance to mount and share it over network?
Best regards,
Nic

RMerl, 2016-10-30:
No, the filesystems are not supported.

closed 2016-10-30 by RMerl

blackfuel, 2016-10-30:
I have tried this on my RT-AC68U and it works.  However, you must compile some kernel modules:
**~/asuswrt-merlin/release/src-rt-6.x.4708/linux/linux-2.6.36/config_base.6a**
CONFIG_ISO9660_FS=m
CONFIG_JOLIET=y
CONFIG_ZISOFS=y
CONFIG_UDF_FS=m
CONFIG_UDF_NLS=y

**EXAMPLE: How to mount a Windows 10 DVD on the router**
/bin/mkdir -p /tmp/media
/sbin/modprobe sr_mod
/bin/mount -t udf -o ro /dev/sr0 /media

**/bin/dmesg**
usb 2-2.3: new high speed USB device using ehci_hcd and address 15
scsi1 : usb-storage 2-2.3:1.0
scsi 1:0:0:0: CD-ROM            MATSHITA DVD-RAM UJ8E0    K801 PQ: 0 ANSI: 0
scsi 1:0:0:0: Attached scsi generic sg1 type 5
sr0: scsi3-mmc drive: 62x/62x writer dvd-ram cd/rw xa/form2 cdda tray
cdrom: Uniform CD-ROM driver Revision: 3.20
sr 1:0:0:0: Attached scsi CD-ROM sr0
UDF-fs: Partition marked readonly; forcing readonly mount

bigpondcableguy, 2016-10-30:
Lets start using our routers as PC's, yeah fantastic idea.......

And the point of mounting a Windows 10 DVD is? Nah don't even explain, I don't even care.

blackfuel, 2016-10-30:
No firmware update is required; it's just loadable kernel modules.

Nic2012, 2016-10-31:
Thanks for the interesting but different answers, the admin says no, others say yes ;)
> However, you must compile some kernel modules:
> ~/asuswrt-merlin/release/src-rt-6.x.4708/linux/linux-2.6.36/config_base.6a
BTW I am not an linux expert, how can I compile the kernel?

RMerl, 2016-10-31:
I'm saying "no" because it won't do it out-of-the-box, and I have no plans to change this.  That doesn't mean that you can't manually compile the missing modules (and pray they don't generate kernel symbol changes, which cause issues with the existing closed source modules).

Nic2012, 2016-10-31:
Ok, as an alternative I could share the CD-ROM by the virtualHere daemon:
https://www.virtualhere.com/usb_server_software
What would be the best place to store this utility persistently on the Merlin distro?

unsynaps, 2016-10-31:
Just ISO the disk or dump its contents to a flash drive. Plug that into the router and share it out with the tools already built in.

blackfuel, 2016-11-01:
How to get the kernel modules for the UDF filesystem.

For RT-AC56U, download this firmware image and use 7-Zip to extract the extra kernel modules.  Copy them to your router's /jffs/bin.
https://github.com/blackfuel/asuswrt-merlin/releases/download/380.62-1/RT-AC56U_380.62_1_blackfuel.zip

Kernel modules for the UDF filesystem:
`lib\modules\2.6.36.4brcmarm\kernel\lib\crc-itu-t.ko
lib\modules\2.6.36.4brcmarm\kernel\fs\udf\udf.ko`

Login to the router and load the kernel modules.
`cd /jffs/bin
insmod crc-itu-t.ko
insmod udf.ko`

If it doesn't work, you may need to extract other kernel modules from the firmware image.  Use the 'dmesg' command to learn the names of any additional kernel modules, or kernel modules for other CD/DVD filesystems.  It's possible to mount an iso9660 filesystem or an ISO image too.  There's also a chance it could break due to different versions of kernel symbols.

Nic2012, 2016-11-05:
I am not able to copy anything into /jffs/bin.
With WinSCP I dont see this directory, although hidden files/dirs are enabled in WinSCP.
But when I try to copy via Putty and cp... I get "cp: can't create '/bin/...': Read-only file system"

blackfuel, 2016-11-06:
The /jffs folder is the router's flash memory area; it's writable and persistent.  The /bin folder is in the firmware image; it's read-only.  So, you may copy the extra kernel modules anywhere on /jffs.  I recommended /jffs/bin because it's not used by the firmware; however, you must create this sub-folder.  Now you have a place to keep the extra kernel modules, without needing to re-flash a new firmware image.

**Example**
`mkdir -p /jffs/bin`

Nic2012, 2016-11-07:
I do not really understand:
As I mentioned on my router there is already a hidden, readonly folder "/jffs/bin".
It will not give any further problems when I create a persistent folder with the same name side by side?

blackfuel, 2016-11-07:
Maybe try putting your extra kernel modules in /tmp/bin instead?  It's writable, but it looses the files when the power to the router is cut.  Now you can test the CD-ROM feature without much trouble?