Asuswrt-Merlin/asuswrt-merlin.legacy #1080: Not compiling when PARENTAL2=n

blackfuel, 2016-10-30:
Here's a patch to fix the issue with 380.63-beta1 not compiling when PARENTAL2=n.
[asuswrt-merlin-380.63-beta1-parental-fix.patch.txt](https://github.com/RMerl/asuswrt-merlin/files/560624/asuswrt-merlin-380.63-beta1-parental-fix.patch.txt)

azdps, 2016-11-04:
I can confirm that there is build failure when attempting to compile when PARENTAL2  is set to PARENTAL=n. I'm using the latest source code from Oct 30, 2016. Your recent change to [firewall.c](https://github.com/RMerl/asuswrt-merlin/commit/f5ea81fffd4c284db417c6f43b20bdf6c69cf47d) must have introduced the issue. Implementing blackfuel's patch or another fix would be great.

RMerl, 2016-11-04:
The proposed patch should fix this, however I want to wait after the 380.63 beta cycle has completed before applying it to the master repo.

commit 3c50783d (2016-11-08): rc: Fix building with parental control disabled (fixes #1080)
closed 2016-11-08 by RMerl