Asuswrt-Merlin/asuswrt-merlin.legacy #1081: AC68U: Dual WAN with failover is broken after upgrade from 380.58 to 380.62_1

imammedo, 2016-10-31:
Screenshot with Dual WAN settings is attached [image].

From both uplinks the router should get IP by DHCP, however when I ssh into it 
'ip addr' shows that vlan3 interface for backup link is down:

  9: vlan3@eth0: <BROADCAST,MULTICAST> mtu 1500 qdisc pfifo state DOWN 
     link/ether 30:5a:3a:c3:f9:38 brd ff:ff:ff:ff:ff:ff

The vlan3 interface by itself seems fine as I can manually put it in up state and then feed it to "udhcpc -i vlan3 -b"  after which it gets IP address from backup uplink.

imammedo, 2016-10-31:
the same issue with 380.61_0 firmware

RMerl, 2016-10-31:
I don't touch the Dual WAN code, as it's too complex, and undocumented.

closed 2016-10-31 by RMerl

pepeEL, 2016-10-31:
Send issue to ASUS