Asuswrt-Merlin/asuswrt-merlin.legacy #1082: AsusWRT-Merlin - secondary WAN disable

chimcen, 2016-10-31:
Hi,

is or will it possible to manually disable (tun off) the secondary WAN? I use the option Dual-WAN because i'm behind a cable modem on LAN1. The router wants to setup a secondary WAN (for fallback e.g.), but i didn't have a secondary WAN connected - so the router will alltime mark the secondary WAN as "nor connection - DHCP problem"...
It not a technicaly problem - its a cosmetic one...
[image]

RMerl, 2016-10-31:
I don't touch the dual wan code, as it's too complex, and undocumented.

closed 2016-10-31 by RMerl

pepeEL, 2016-10-31:
Send issue to ASUS