Asuswrt-Merlin/asuswrt-merlin.legacy #1085: DNS server returns 10.0.0.1 for any LAN hostname when unable to connect to internet

Coaxial, 2016-11-04:
Our ISP suffers an outage at the moment, and the router can't connect to the internet.

I have several hosts on my LAN, for which the router's DHCP assigns a fixed IP.

If I try to resolve `myhostname`, instead of returning the IP it should (192.168.1.xxx), it returns 10.0.0.1. Same thing for any other hostname on the LAN, it returns 10.0.0.1. I can still access the hosts by their IP.

Happy to provide more info as needed.

```
$ dig mainframe

; <<>> DiG 9.8.3-P1 <<>> mainframe
;; global options: +cmd
;; Got answer:
;; ->>HEADER<<- opcode: QUERY, status: NOERROR, id: 29954
;; flags: qr aa rd ra; QUERY: 1, ANSWER: 1, AUTHORITY: 0, ADDITIONAL: 0

;; QUESTION SECTION:
;mainframe.    			IN     	A

;; ANSWER SECTION:
mainframe.     		0      	IN     	A      	10.0.0.1

;; Query time: 3 msec
;; SERVER: 192.168.1.1#53(192.168.1.1)
;; WHEN: Thu Nov  3 21:29:26 2016
;; MSG SIZE  rcvd: 43
```

```
$ nslookup mainframe
Server:		192.168.1.1
Address:       	192.168.1.1#53

Name:  	mainframe.lan
Address: 10.0.0.1
```

Coaxial, 2016-11-04:
Now that the ISP fixed the fibre, this is what I'm getting (no router reboot or config change, the only difference is that the fibre optics got fixed on the ISP's side)

```
$ dig mainframe

; <<>> DiG 9.8.3-P1 <<>> mainframe
;; global options: +cmd
;; Got answer:
;; ->>HEADER<<- opcode: QUERY, status: NOERROR, id: 51743
;; flags: qr aa rd ra; QUERY: 1, ANSWER: 1, AUTHORITY: 0, ADDITIONAL: 0

;; QUESTION SECTION:
;mainframe.    			IN     	A

;; ANSWER SECTION:
mainframe.     		0      	IN     	A      	192.168.1.129

;; Query time: 4 msec
;; SERVER: 192.168.1.1#53(192.168.1.1)
;; WHEN: Fri Nov  4 09:15:32 2016
;; MSG SIZE  rcvd: 43
```

```
$ nslookup mainframe
Server:		192.168.1.1
Address:       	192.168.1.1#53

Name:  	mainframe.lan
Address: 192.168.1.129
```

RMerl, 2016-11-04:
Try disabling the web redirection setting under Administration -> System.

Coaxial, 2016-11-04:
Thanks, I'll do that.

closed 2016-11-06 by RMerl