Asuswrt-Merlin/asuswrt-merlin.legacy #1086: RFE: Sort DHCP Manual Assignment

leoslion, 2016-11-04:
Description of problem:
You can manually assign/map MAC addresses to ip addresses and hostnames. It allows up to 128 assignments. it would be nice to be able to sort and filter them.

Version-Release number of selected component (if applicable):
all versions. I'm specifically using 38.57

How reproducible:
everytime

Steps to Reproduce:
1. login to admin portal
2. Goto Lan -> DHCP Server
3. Set "Enable manual assignment" to yes
4. reserve at least 2 mac addresses to ip and hostnames
5. click on "IP Address" or "Hostname"

Actual results:
Nothing

Expected results:
sorting ascending and descending of "ip addresses" or "hostname"

Additional info:

fanaticDavid, 2016-11-04:
This already works for me. Have you tried sorting the list **after** applying the changes?

leoslion, 2016-11-05:
Apparently sorting was added and I didn't see it in the release notes. Filtering is not there but sorting is the biggest issue for me. Thanks!

closed 2016-11-05 by leoslion