Asuswrt-Merlin/asuswrt-merlin.legacy #1095: Policy based routing fix for Dual WAN setups

gunnyst, 2016-11-09:
Dear Sirs,

I need to use an OpenVPN tunnels together with a Dual WAN setup.
My clients where unable to use the tunnel even when I manually added policy based routing.

The issue seems to be the ip rule priorities. In a Dual WAN setup with:
(1) load balancing
(2) manual routes
(3) one policy based route for the VPN
you may easily stumble into this issue kind of result:

```
0:	from all lookup local
100:	from 192.168.0.10 lookup wan0
100:	from 192.168.0.11 lookup wan0
100:	from 192.168.0.20 lookup wan1
100:	from 192.168.0.21 lookup wan1
100:    ... some more 100-rules here ...
200:    ... some more 200-rules here ...
400:    ... some more 400-rules here ...
10301:	from 192.168.0.10 to 172.20.1.0/24 lookup ovpnc1
32766:	from all lookup main
32767:	from all lookup default
```

The issue with this will be that the 100-rule for 192.168.1.10 will be applied before our policy based 10301-rule (which would be more specific since it also provides a destination!

IMHO, there are various fixes for this:

(1) document it telling people they CANNOT use manual routes in a dual WAN setup together with policy based routes
(2) fix A: add the possibility to choose both "Primary WAN" + "Secondary WAN" interfaces in case of policy base routes (instead of just "WAN") and ALSO hint at removing manual routes from Dual WAN setups as a consequence
(3) fix B: add policy based routes with higher priority (equals a lower number then "100" ie. "50")

Here is what you can manually do on the command line (SSH) to fix it ad-hoc or using a manual script:

```
ip rule del prio 10301 from 192.168.0.10 to 172.20.1.0/24 lookup ovpnc1
ip rule add prio 50 from 192.168.0.10 to 172.20.1.0/24 lookup ovpnc1
```

... which will result in the following rule-set and a working configuration ...

```
0:	from all lookup local
50:	from 192.168.0.10 to 172.20.1.0/24 lookup ovpnc1
100:	from 192.168.0.10 lookup wan0
100:	from 192.168.0.11 lookup wan0
100:	from 192.168.0.20 lookup wan1
100:	from 192.168.0.21 lookup wan1
200:    ... some 200-rules here ...
400:    ... some 400-rules here ...
32766:	from all lookup main
32767:	from all lookup default
```

.... where Traffic from 192.168.0.10 will by default go out through the Primary WAN, but in case you try reaching things on the other side of your VPN (172.20.1.0/24 here), traffic will go through your VPN tunnel as expected.

Hope this helps (and someone can implement it).... well truth be said I'd love to put my hands on the code, but I'm new here and would rather not risk messing things up :-D

Cheers

Günther

gunnyst, 2016-11-09:
This is a working solution for the current situation (asuswrt-merlin 380.63).

```
#!/bin/sh
#      
# filename: /jffs/scripts/openvpn-event
#                                                                                                                                
                                                                                                                                 
sleep 1                                                                                                                          
                                                                                                                                 
RULES="to 172.20.1.0/24 lookup ovpnc1;to 10.1.0.0/16 lookup ovpnc2"
IFS=";"                                                                  
                                                                         
for RULE in $RULES                                                       
do                                                                       
  if [ ! -z "$RULE" -a "`ip rule list | grep -c \"$RULE\"`" = "0" ]; then
    sh -c "ip rule add prio 50 $RULE"
  fi  
done  
      
exit 0
```

Just remember to set _Redirect Internet traffic_ to _Policy Rules_ in order to have ASUSWRT automatically create the routing tables for you!

RMerl, 2016-11-10:
Why not simply move the dualwan static routes from a base pri of 100/200 to a base pri of 20100/20200?  This would be a much simpler fix (it's just two defines that need updating in the code).

Moving the OpenVPN rules down is not an option.  Each client need to handle up to 200 rules  (100 wan and 100 vpn rules), times 5 clients (so a total of 1000 slots must be reserved).

RMerl, 2016-11-10:
Actually, more than just a pair of defines, I see that other parts of rc/wan.c are using inline priorities of 100, 200, 300 and 400 rather than rely on defines within add_multi_routes().  Still seem fairly trivial to change after a quick glance (a more in depth analysis will need to be made before making such a change, just to be sure.)

gunnyst, 2016-11-10:
Hi there!

I was already wondering why the static routes had such low priority numbers when the maximum is 32767 :-P

Just one note (and maybe I'm wrong): why would you have to handle 100 wan + 100 vpn rules, all of them with a single/different priority? One priority (50, 100, 200, 300, 400) may contain a thousand single rules pointing at the 5 tables we currently have (from ovpnc1 to ovpnc5 - one for each VPN client). There should be no need to increase priority IDs just because of a different WAN/VPN (that way routing priorities would be handed out by order of creation which is somewhat confusing and for people who don't know better easily looks like "random behaviour"). IMHO, priority IDs are just what the say they are: priorities.

**But!** I do heavily agree with you that it would be better to have:

1. higher rule priorities for the default routing tables (currently 100/200/300/400)
2. have those priorities farther apart then they are now (since we are at it: what about 20000/21000/22000/23000?)
3. have policy rules on a lower priority number (by their nature they should be "more specific" - lower priority number equals higher priority, so.... say 10000?)

Thanks!

Günther

RMerl, 2016-11-10:
When I researched it, there was no official documentation stating that it was OK to have multiple rules share a same priority.  A few second-hand references were going both ways, stating they should be unique, or stating they could be shared.  So, I went with the safest bet, and made them all use unique IDs.

Policy rules are already starting at 10000 (changed from the original value of 1000, as some users complained about the lack of available slots in the higher prio range).

I'll analyze the rest of the routing code first, but I will most likely bump the regular static routes into the 20000 range.

gunnyst, 2016-11-10:
Hi Eric,

you're right, I just had a look at the ip rule manual page and it states:

```
              priority PREFERENCE
                     the priority of this rule.  PREFERENCE is an unsigned
                     integer value, higher number means lower priority.
                     Each rule should have an explicitly set unique priority
                     value.  The options preference and order are synonyms
                     with priority.
```

This is strange since it works quite well (the asus router is running with those ripetitive 100-, 200- and 400-rules I stated in my first post for 3 days now) and gives no warning when you use it that way.

Leaving aside the issue about this being more or less intuitive I follow along with your interpretation about sticking to the documented solution.

Sorry about that 10000 thing... I did overlook that it was already that way!

Thanks again

Günther

RMerl, 2016-11-14:
Fixed with 6b9c2bccb74d0cf336e47c16e7ee96aa672c5c4f .

closed 2016-11-14 by RMerl