Asuswrt-Merlin/asuswrt-merlin.legacy #1096: ssh-ed25519 auth support

bipoman, 2016-11-09:
Request for ssh-ed25519 authentication support please

RMerl, 2016-11-09:
Curve25519 is already enabled in Dropbear:

https://github.com/RMerl/asuswrt-merlin/blob/master/release/src/router/dropbear/options.h#L165

If you're talking about some other variant, then it's not supported by Dropbear.

closed 2016-11-09 by RMerl

bipoman, 2016-11-10:
Eric,

Thanks for the quick reply and link.  I see dropbear supports it, but how do I setup the ssh authentication key in the Merlin gui to except my id_ed25519.pub key that I generated on my Mac using the following command:  ssh-keygen -o -a 100 -t ed25519

I also tried to ssh-copy-id <username>@<router ip>  which copied the id to /root/.ssh/authorized_keys and then cp /root/.ssh/authorized_keys /etc/dropbear/ and then chmod 700 /etc/dropbear and chmod 600 /etc/dropbear/authorized_keys

Please, if you can provide any insight into how I can make this work, step by step would be nice ☺, I would really appreciate it.

Thanks,

Daryl

https://github.com/RMerl/asuswrt-merlin/blob/master/release/src/router/dropbear/options.h#L165

RMerl, 2016-11-10:
Paste it on the SSH key field under Administration -> System on the webui.

bipoman, 2016-11-11:
Eric,

I have already tried pasting in the key:

[cid:image001.png@01D23B6D.4B3BCCD0]

However, when I try to ssh to the router I get:

Permission denied (publickey).

Thoughts?

Thanks,

Daryl

bipoman, 2016-11-11:
Eric,

Also this is the debug of me trying to ssh:

BELC02RN4UEG8WN:~ dcarr$ ssh -vvvv syslocal@securenet-rtr
OpenSSH_7.2p2, LibreSSL 2.4.1
debug1: Reading configuration data /etc/ssh/ssh_config
debug1: /etc/ssh/ssh_config line 20: Applying options for *
debug2: resolving "securenet-rtr" port 22
debug2: ssh_connect_direct: needpriv 0
debug1: Connecting to securenet-rtr [192.85.15.1] port 22.
debug1: Connection established.
debug1: key_load_public: No such file or directory
debug1: identity file /Users/dcarr/.ssh/id_rsa type -1
debug1: key_load_public: No such file or directory
debug1: identity file /Users/dcarr/.ssh/id_rsa-cert type -1
debug1: key_load_public: No such file or directory
debug1: identity file /Users/dcarr/.ssh/id_dsa type -1
debug1: key_load_public: No such file or directory
debug1: identity file /Users/dcarr/.ssh/id_dsa-cert type -1
debug1: key_load_public: No such file or directory
debug1: identity file /Users/dcarr/.ssh/id_ecdsa type -1
debug1: key_load_public: No such file or directory
debug1: identity file /Users/dcarr/.ssh/id_ecdsa-cert type -1
debug1: identity file /Users/dcarr/.ssh/id_ed25519 type 4
debug1: key_load_public: No such file or directory
debug1: identity file /Users/dcarr/.ssh/id_ed25519-cert type -1
debug1: Enabling compatibility mode for protocol 2.0
debug1: Local version string SSH-2.0-OpenSSH_7.2
debug1: Remote protocol version 2.0, remote software version dropbear_2016.74
debug1: no match: dropbear_2016.74
debug2: fd 3 setting O_NONBLOCK
debug1: Authenticating to securenet-rtr:22 as 'syslocal'
debug3: hostkeys_foreach: reading file "/Users/dcarr/.ssh/known_hosts"
debug3: record_hostkey: found key type ECDSA in file /Users/dcarr/.ssh/known_hosts:1
debug3: load_hostkeys: loaded 1 keys from securenet-rtr
debug3: hostkeys_foreach: reading file "/etc/ssh/ssh_known_hosts"
debug3: order_hostkeyalgs: prefer hostkeyalgs: ecdsa-sha2-nistp256-cert-v01@openssh.com,ecdsa-sha2-nistp384-cert-v01@openssh.com,ecdsa-sha2-nistp521-cert-v01@openssh.com,ecdsa-sha2-nistp256,ecdsa-sha2-nistp384,ecdsa-sha2-nistp521
debug3: send packet: type 20
debug1: SSH2_MSG_KEXINIT sent
debug3: receive packet: type 20
debug1: SSH2_MSG_KEXINIT received
debug2: local client KEXINIT proposal
debug2: KEX algorithms: curve25519-sha256@libssh.org,ecdh-sha2-nistp256,ecdh-sha2-nistp384,ecdh-sha2-nistp521,diffie-hellman-group-exchange-sha256,diffie-hellman-group-exchange-sha1,diffie-hellman-group14-sha1,ext-info-c
debug2: host key algorithms: ecdsa-sha2-nistp256-cert-v01@openssh.com,ecdsa-sha2-nistp384-cert-v01@openssh.com,ecdsa-sha2-nistp521-cert-v01@openssh.com,ecdsa-sha2-nistp256,ecdsa-sha2-nistp384,ecdsa-sha2-nistp521,ssh-ed25519-cert-v01@openssh.com,ssh-rsa-cert-v01@openssh.com,ssh-ed25519,rsa-sha2-512,rsa-sha2-256,ssh-rsa
debug2: ciphers ctos: chacha20-poly1305@openssh.com,aes128-ctr,aes192-ctr,aes256-ctr,aes128-gcm@openssh.com,aes256-gcm@openssh.com,aes128-cbc,aes192-cbc,aes256-cbc,3des-cbc
debug2: ciphers stoc: chacha20-poly1305@openssh.com,aes128-ctr,aes192-ctr,aes256-ctr,aes128-gcm@openssh.com,aes256-gcm@openssh.com,aes128-cbc,aes192-cbc,aes256-cbc,3des-cbc
debug2: MACs ctos: umac-64-etm@openssh.com,umac-128-etm@openssh.com,hmac-sha2-256-etm@openssh.com,hmac-sha2-512-etm@openssh.com,hmac-sha1-etm@openssh.com,umac-64@openssh.com,umac-128@openssh.com,hmac-sha2-256,hmac-sha2-512,hmac-sha1
debug2: MACs stoc: umac-64-etm@openssh.com,umac-128-etm@openssh.com,hmac-sha2-256-etm@openssh.com,hmac-sha2-512-etm@openssh.com,hmac-sha1-etm@openssh.com,umac-64@openssh.com,umac-128@openssh.com,hmac-sha2-256,hmac-sha2-512,hmac-sha1
debug2: compression ctos: none,zlib@openssh.com,zlib
debug2: compression stoc: none,zlib@openssh.com,zlib
debug2: languages ctos:
debug2: languages stoc:
debug2: first_kex_follows 0
debug2: reserved 0
debug2: peer server KEXINIT proposal
debug2: KEX algorithms: curve25519-sha256@libssh.org,ecdh-sha2-nistp521,ecdh-sha2-nistp384,ecdh-sha2-nistp256,diffie-hellman-group14-sha1,diffie-hellman-group1-sha1,kexguess2@matt.ucc.asn.au
debug2: host key algorithms: ecdsa-sha2-nistp521,ssh-rsa,ssh-dss
debug2: ciphers ctos: aes128-ctr,aes256-ctr,aes128-cbc,aes256-cbc,twofish256-cbc,twofish-cbc,twofish128-cbc,3des-ctr,3des-cbc
debug2: ciphers stoc: aes128-ctr,aes256-ctr,aes128-cbc,aes256-cbc,twofish256-cbc,twofish-cbc,twofish128-cbc,3des-ctr,3des-cbc
debug2: MACs ctos: hmac-sha1-96,hmac-sha1,hmac-sha2-256,hmac-sha2-512
debug2: MACs stoc: hmac-sha1-96,hmac-sha1,hmac-sha2-256,hmac-sha2-512
debug2: compression ctos: none
debug2: compression stoc: none
debug2: languages ctos:
debug2: languages stoc:
debug2: first_kex_follows 0
debug2: reserved 0
debug1: kex: algorithm: curve25519-sha256@libssh.org
debug1: kex: host key algorithm: ecdsa-sha2-nistp521
[... 57 lines cut ...]

Thanks,

Daryl

From: Daryl Carr <daryl.carr@outlook.com>
Date: Thursday, November 10, 2016 at 4:12 PM
To: RMerl/asuswrt-merlin <reply@reply.github.com>
Subject: Re: [RMerl/asuswrt-merlin] ssh-ed25519 auth support (#1096)

Eric,

I have already tried pasting in the key:

[cid:image001.png@01D23B6F.9E2E78A0]

However, when I try to ssh to the router I get:

Permission denied (publickey).

Thoughts?

Thanks,

Daryl

RMerl, 2016-11-11:
Apparently, ed25519 and curve25519 are not the same thing.  Of the two, Dropbear only supports curve25519, so you will have to switch to a supported algorithm.

bipoman, 2016-11-11:
How can I generate a curve25519 key?

Thanks for all your help.

Daryl

RMerl, 2016-11-11:
No idea. I've never heard of curve25519 before.

themiron, 2016-11-11:
To be clear, curve25519 is key exchange algo and is supported by dropbear. ed25519 is signature algo, which is not supported by dropbear (yet). That's why you can use ssh session encryption based on 25519, but you can't use ed25519 key-based client auth.

bipoman, 2016-11-11:
Thanks Vladislav for the explanation.  What do you recommend then for the most secure ssh connection method to my router from my Mac and can provide me the needed instructions of generating the key?

Thanks,

Daryl