Asuswrt-Merlin/asuswrt-merlin.legacy #1099: RT-AC66U_B1 router does not allow firmware update

gorantornqvist, 2016-11-10:
Hi,
Trying to flash RT-AC66U_380.63_0.trx on my new RT-AC66U_B1 router but I´m getting the following message:

Invalid Firmware Upload

To comply with regulatory amendments, we have modified our certification rule to ensure better firmware quality. This version is not compatible with all previously released ASUS firmware and uncertified third party firmware. Please check our official websites for the certified firmware.

Any way around this?

stevek123, 2016-11-10:
see http://www.snbforums.com/threads/can-asus-rt-ac66u-b1-run-asuswrt-merlin.35323/

gorantornqvist, 2016-11-11:
Thanks ...so the AC68U firnware *should* work according to the thread. Guess I will have to try .

mrath, 2016-11-12:
I have the same question, and have yet to get an answer from this page, or the posted link.
Anything we can do to help include support for the B1/+ model?

RMerl, 2016-11-12:
This tracker is for bug reports.  Please use the support forums at SNBForums for support requests.

closed 2016-11-12 by RMerl

dandv, 2017-04-22:
I no longer see this issue on my AC68U with RT-AC68U_380.65_4.trx. It's flashing as I typ