Asuswrt-Merlin/asuswrt-merlin.legacy #1105: Block routed clients if tunnel goes down not working on RT-AC88U

nbari, 2016-11-13:
When using the OpenVPN client, the  ``Redirect Internet traffic`` -> ``Policy Rules`` works as expected but if selected the option ``Block routed clients if tunnel goes down`` it is not working.

If the VPN goes down the clients specified in the policy rules

As a workaround selecting the option ``Firewall`` -> ``Custom`` and adding the following in to ``/jffs/scripts/firewall-start`` works

```
#!/bin/sh

sleep 4

iptables -I FORWARD -i br0 -o tun11 -j ACCEPT
iptables -I FORWARD -i tun11 -o br0 -j ACCEPT
iptables -I FORWARD ! -o tun11 -s 192.168.1.xxx -j DROP
iptables -I FORWARD ! -o tun11 -s 192.168.1.xxx -j DROP
iptables -I INPUT -i tun11 -j REJECT
iptables -t nat -A POSTROUTING -o tun11 -j MASQUERADE
```

Using firmware 380.63_2

nbari, 2016-11-13:
I notice that every time I change the parameters like ``Accept DNS Configuration`` , the rules stack instead of been create from new, so toggling between options ``strict``, ``explicit``  I ended with a ``client1-dns.sh`` file like this:

```
admin@RT-AC88U-E3C0:/tmp/etc/openvpn/fw# cat client1-dns.sh
iptables -t nat -N DNSVPN1
iptables -t nat -N DNSVPN1
iptables -t nat -A DNSVPN1 -s 192.168.1.100 -j DNAT --to-destination 198.18.0.1
iptables -t nat -A DNSVPN1 -s 192.168.1.13 -j DNAT --to-destination 198.18.0.1
iptables -t nat -A PREROUTING -p udp -m udp --dport 53 -j DNSVPN1
iptables -t nat -A PREROUTING -p tcp -m tcp --dport 53 -j DNSVPN1
iptables -t nat -N DNSVPN1
iptables -t nat -N DNSVPN1
iptables -t nat -N DNSVPN1
iptables -t nat -N DNSVPN1
iptables -t nat -N DNSVPN1
iptables -t nat -A DNSVPN1 -s 192.168.1.100 -j DNAT --to-destination 198.18.0.1
iptables -t nat -A DNSVPN1 -s 192.168.1.13 -j DNAT --to-destination 198.18.0.1
iptables -t nat -A PREROUTING -p udp -m udp --dport 53 -j DNSVPN1
iptables -t nat -A PREROUTING -p tcp -m tcp --dport 53 -j DNSVPN1
iptables -t nat -N DNSVPN1
iptables -t nat -N DNSVPN1
iptables -t nat -N DNSVPN1
iptables -t nat -N DNSVPN1
iptables -t nat -N DNSVPN1
iptables -t nat -N DNSVPN1
iptables -t nat -N DNSVPN1

```

Clearing the ``fw`` files and stoping/staring the vpn seem to be helping.

RMerl, 2016-11-25:
Works properly for me.  When the DNS configuration is changed,the VPN client gets stopped then restarted, which causes the updown script to be called with a "down" event.  The script will flush the chain, and delete any existing client1-dns.sh.  Make sure you're not preventing the normal updown.sh script from running on both up and down events.  That could also explain why the routing tables aren't updated either on the down event.

closed 2016-11-25 by RMerl

nbari, 2016-11-25:
@RMerl could you please explain, give a hint about how to make sure the normal updown.sh script is up running on both up and down events?

thanks in advance