Asuswrt-Merlin/asuswrt-merlin.legacy #1107: Support for RT-AC66U hardware revision B1?

peterdenk, 2016-11-15:
On the site (http://asuswrt.lostrealm.ca/about)
RT-AC66U (hardware revision B1 is NOT supported)

In README.md
RT-AC66U

Which one is correct?

RMerl, 2016-11-15:
Hardware revision B1 is a newer revision that only appeared on the 
market a few weeks ago.  That new revision is not supported.

oldskooler, 2016-11-19:
Is there any plan in the near future that you might do support this device?
Thank you very much

RMerl, 2016-11-19:
I don't know, Asus didn't answer me when I asked if the hardware was 100% compatible with the RT-AC68U firmware or not.  If it is, then people can simply flash the RT-AC68U firmware.  If not, then I don't know if it will ever be supported or not, as it will depend on the amount of work required versus how widespread this model will be.

peterdenk, 2016-11-20:
Ok, then README.md needs to be updated with this information.

commit bd44afd7 (2016-11-21): docs: Refer to the Wiki for the up-to-date list of supported models (closes #1107)
closed 2016-11-21 by RMerl