Asuswrt-Merlin/asuswrt-merlin.legacy #1109: native IPv6 broken in 380.63_2

tuxpeople, 2016-11-17:
This may be related to #936 but not to #999 as I do not have SSH enabled.

IPv6 is not working (used to get a IPv6 Prefix via DHCPv6 Prefix Delegation). Not sure when it stopped working, but it worked not long ago.

I understand you do not have the possibility to test native IPv6. I may assist you, as I have native IPv6 and I work in my ISP's NOC ;)

I do not recieve the prefix, the IPv6 Screen is empty and there is no IPv6 neighbor visible on the other (ISP) end of the internet connection.

Settings:
Connection type	native
DHCP-PD	enable
Release prefix on exit	enable

There was no configuration change on my ASUS (which is, in case this matters, an RT-AC68U). I tried to torn off and on IPv6 again, but without luck.

Please let me know if I can provide any additional information.

tuxpeople, 2016-11-18:
It's working again after downgrading to Firmware 380.62_1

RMerl, 2016-11-25:
I don't touch IPv6 anymore.  Any issue will have to be resolved upstream - I have no way of testing this properly.

closed 2016-11-25 by RMerl

geeohgeegeeoh, 2017-04-14:
Native IPv6 via dhcp-pd is broken for Australian NBN fiber on 380.65, the layer3 is skymesh. Works fine in 380.62_1

Seen some suggestions this is a default ip6tables rule problem. I'm a bsd person not a Linux person if somebody can tell me a simple way to get rid of any ip6tables blocks to see if that helps I'm happy to try.