Asuswrt-Merlin/asuswrt-merlin.legacy #1110: System Log - General Log are not sorted by time ..

lukasz-karolewski, 2016-11-18:
entries in the System Log - General Log are not sorted by time ..  this makes reading very difficult.

it's really weird but it's partially sorted, there are sorted batches that are then mixed seemingly in random order

part of the problem be the date format Nov 17 23:54:09 this will never get sorted correctly, 

is it possible to change time format to iso8601? yyyy-mm-ddT23:54:09 ?

RMerl, 2016-11-18:
The system log is not a database, it's a flat file.  Entries cannot be sorted.

closed 2016-11-18 by RMerl

lukasz-karolewski, 2016-11-19:
changing syslog format to use iso would be enough 

optionally sorting on the view is a one liner

contents = content.split('\n).sort().join('\n')

RMerl, 2016-11-19:
Busybox's date format is not configurable.