Asuswrt-Merlin/asuswrt-merlin.legacy #1113: Can't Update

okyere, 2016-11-19:
I'm new to all this. My RT-AC68U is on firmware 376.47 (Merlin build) after following a tutorial on youtube. I've been trying to upgrade to the latest build but no luck. When I tried to update to the latest version by uploading under Administration->Firmware upgrade. It uploads successfully but then the page returns and it is still the older firmware (376.47). Same thing happens when I try to update with tftp with rescue mode. It is successful but when I log back in, it still shows the older version. I've also tried the firmware restoration tool. It uploads up to 100% and then for some reason it stops and says that it has disconnected. When I look at the router at that time, it looks like has is rebooting. It comes back on and still has the same older firmware. Am I doing something wrong? Thanks.

marodor, 2016-11-19:
Cfe bootloader probly need to be updated...
Flash with 378.55 first to upgrade CFE and flash with the latest after

okyere, 2016-11-19:
That did it! Thank you.

closed 2016-11-19 by okyere