Asuswrt-Merlin/asuswrt-merlin.legacy #1115: Native IPV6 via PPPoe does not set default gateway

Netriak, 2016-11-21:
When I configure native IPV6 on my pppoe interface (ppp0), the default gateway is not set.
That is, the "WAN IPv6 Gateway" field on the ipv6 tab of the system log page is empty, and there is no IPV6 default route at all.
Naturally, this means that reaching any host outside of the local network via IPV6 is impossible.

Manually adding the proper default route (to the link remote local interface provided to pppd, which is visible in the system log) via SSH/wan-start script restores full IPV6 funtionality:

route add -A inet6 default gw {remote link local address goes here} dev ppp0

That is, the "WAN IPv6 Gateway" is properly filled in and non-local hosts are accessible via IPV6.

using firmware 380.63_2

RMerl, 2016-11-25:
I don't touch IPv6 anymore.  Any issue will have to be resolved upstream - I have no way of testing this properly.

closed 2016-11-25 by RMerl

Cilph, 2017-05-03:
Thank you @Netriak. Thanks to you I was able to figure out why my connection wasn't working.