Asuswrt-Merlin/asuswrt-merlin.legacy #1116: QoS sucks while uploading

ryannathans, 2016-11-21:
When another device on the network is uploading, I experience horrid latency issues (>2000ms on a ping). Router is RT-AC5300.

I've tried adaptive QoS, traditional QoS and bandwidth limiting. The only thing that "solves" the problem is bandwidth limiting upload to 0.6Mb which is really non-ideal. I'm on cable and my speeds vary a lot.

Offpeak 100Mb/s down, 2Mb/s up
Peak 12Mb/s down. 0.9Mb/s up (or worse)

For my test, I have a phone on WiFi uploading a file to Mega. It looks like web browsing in QoS statistics.

My PC is doing little more than "ping 8.8.8.8 /t". The attached screenshot is with adaptive QoS set to custom, with the order Gaming, Video, VoIP, Web, File Transfer, Others. Bandwidth setting is on automatic. The results are roughly identical unless I use the bandwidth limiting to limit the device to 0.6/0.7Mb/s. My current max upload speed is around 1.2Mb/s. I've tried all the Queue Discipline modes too.

Probable issue or have I missed something?

[image]

relder001, 2016-11-21:
Ping (ICMP) is a lowest priority protocol, you'll have to use another tool/method to measure and observe simultaneous flows.  Have you seen/tried the HTML5 network test tool '@sourceforge dot net slash speedtest/ ?

RMerl, 2016-11-21:
Please use the support forums at SNBForums for support discussions.

closed 2016-11-21 by RMerl