Asuswrt-Merlin/asuswrt-merlin.legacy #1118: Fix the unsafe TOR

blackfuel, 2016-11-22:
Do you think that using the current Asus TOR integration is unsafe?  Could I be attacked at home?  If yes, here's a patch for your consideration.  I want to use TOR correctly so it does not leak my real IP address.  In some parts of the world these governments will come, track you down, kidnap you, or worse.

So, I am a satisfied Asus customer and I'm happy to fix the issue.  My patch correctly configures the firewall to use TOR safely.  Now you can redirect any IP addresses or IP ranges, in addition to the MAC addresses.

**Example without WebUI**
`/usr/sbin/nvram set Tor_enable=1`
`/usr/sbin/nvram set Tor_redir_list="<aa:bb:cc:dd:ee:ff<192.168.1.222<192.168.1.240-245"`
`#/usr/sbin/nvram unset Tor_redir_list`
`/usr/sbin/nvram commit`
`/sbin/service restart_firewall`

**How to patch the source code**
`cd ~/asuswrt-merlin`
`patch -p2 -i ~/asuswrt-merlin-380.63_2-tor-fix.patch.txt`

[asuswrt-merlin-380.63_2-tor-fix.patch.txt](https://github.com/RMerl/asuswrt-merlin/files/607366/asuswrt-merlin-380.63_2-tor-fix.patch.txt)

RMerl, 2016-11-25:
What specific issue does this patch address?

blackfuel, 2016-11-25:
I think the firewall should be routing all TCP ports through the TOR network.  However, it's only doing TCP ports 80 and 443.  AND, for all other protocols I think it should be blocking UDP and ICMP from reaching the Internet.  From the perspective of a TOR user who invests both time and money to properly maintain her anonymity, it's a critical security issue.

Lastly, I added the ability to redirect IP addresses and IP ranges, for convenience.  The "address checker" code was borrowed from the QOS bandwidth limiter.

blackfuel, 2016-11-26:
A scripting solution to address the unsafe TOR issue?
http://www.snbforums.com/threads/tor-for-all-traffic-from-a-mac.35848/#post-292383

RMerl, 2016-11-26:
I just need to find the time to review your patch before committing it - your description makes sense to me.

commit 1c42870e (2016-12-01): tor: route other ports than just 80/443 through Tor, and block UDP/ICMP traffic (Closes #1118)
closed 2016-12-01 by RMerl