Asuswrt-Merlin/asuswrt-merlin.legacy #1122: High Load  Freezes httpds

acornspaul, 2016-11-25:
This problem only exists when web administration is done via https.  Frequently, when browsing web administration pages using https the page will freeze and not render.  If one duplicates the page using Chromes duplicate page feature the page un-jams and renders.  If you set  up a shell connection to the router (I have an RT-AC68U) and run top you will see the httpds process hog one cpu while things are jammed and then its load goes back down on the next page request (the page duplication).   This makes https unusable and is a security issue.

RMerl, 2016-11-25:
Known issue which appeared in Chrome a few versions ago.  Chrome itself will also peg the CPU during that time.  Can't track the cause, and since it only happens in Chrome (and started following a Chrome update), this isn't something I can fix.

closed 2016-11-25 by RMerl

acornspaul, 2017-01-12:
This appears to be fixed as of 380.64_2