Asuswrt-Merlin/asuswrt-merlin.legacy #1123: REQ: Guest internet wired port (or VLAN tagging)

svrooij, 2016-11-25:
I would like to have the guest internet access (wireless and wired) in a separate VLAN (with a separate ip range).
A toggle with port 4 guest access or something like that.

To take this idea even further it would be great to put traffic with a certain VLAN ID in this guest lan. That way I could provide wired internet from a managed switch to guest. So only internet (with(out) speedlimit) and no lan access.

I found a site that describes this with a bunch of scripts... but a settings pages would be great!

Dannyzen, 2016-11-25:
+1 here. I have some devices wired in that I'd like to isolate off my LAN as well. 

@svrooij would you mind sharing the links if this request is closed?

svrooij, 2016-11-25:
@Dannyzen I'll share it anyway. http://www.snbforums.com/threads/wap-guest-ssid-port-based-vlan.12750/#post-81447

RMerl, 2016-12-01:
Implementing VLAN support would require too many extensive changes throughout the firmware, and would be beyond the scope of this project (it would make it very difficult to stay in sync with Asus).

Asus started work on adding VLAN support a few months ago, we'll have to wait and see if they finalize that support at some point.

closed 2016-12-01 by RMerl