Asuswrt-Merlin/asuswrt-merlin.legacy #1127: only vpn client 1,2 work as a failover.

nbari, 2016-12-01:
When having more than 2 openvpn clients connected, only the first 2 route the traffic.

Having all vpn's connected, seems that the client 1 is the default route, if vpn 1 gets disconnected all traffic goes out using vpn client 2, basically client 2  behaves like a failover for vpn 1 which works very nice.

When disconnecting client 2, I expected the vpn 3 to became the failover and start routing traffic, but notice that either vpn 4, vpn 5 don't route traffic.

I used the configuration of vpn client 3, 4 and 5 on vpn client 2 to test if something was wrong but in all cases, seems that the failover is only working for vpn 1 and 2.

Is there a way to make vpn client 3, 4 5 to behave like failover ?

RMerl, 2016-12-07:
Are all tunnels using totally separate subnets?

nbari, 2016-12-07:
Yes, If I rotate them, for example, vpn 4 5 to be now 1, 2 they work like expected, I mean when disconnecting vpn 1, vpn 2 will behave like failover, but when disconnecting vpn 2, vpn 3 seems (that used to be 1 before) now will not work.

This are the IP's obatined per client:
vpn1  10.10.X.X
vpn2  10.200.X.X
vpn3  10.30.X.X
vpn4 172.16.X.X
vpn5 172.22.X.X

RMerl, 2017-01-01:
A number of timing issues at shutdown were fixed on Git, please re-test with a 380.65 test build.

RMerl, 2017-01-19:
Closing this due to lack of feedback.

closed 2017-01-19 by RMerl