Asuswrt-Merlin/asuswrt-merlin.legacy #1129: OpenVPN Client - Certificate field too short

siborg, 2016-12-02:
Current Limit is set : 3499 characters per field. 

My VPN provider recently issued new certificates. The whole cert is 4568 characters long. Can the field capacity be expanded? 

Trying to upload openvpn config files with such cert can reboot the router and causes additional weird behaviour such us:

Once such config is loaded:

1. Not always but often the router reboots
2. When the router does not reboot the configuration hangs and either drops you back to logon prompt or hangs on applying screen. the new cert file is written to /jffs/openvpn/ folder but no other parts of the config are parsed. 
3. You cannot apply any settings nor change the openvpn client in the web configuration. 

The behaviour returns to normal after deleting the cert file in /jffs/openvpn folder.

RMerl, 2016-12-02:
The current limit is large enough to accommodate a 4096-bit certificate, which is already overkill for home usage (especially for the CPU found in these routers).  Increasing it would require changing numerous buffers throughout the firmware, and would also waste a lot of nvram - not gonna happen, sorry.

closed 2016-12-02 by RMerl

siborg, 2016-12-02:
Thanks. I don't think the actuall key is larger than 4096. Its just the whole <ca> block is a compound cert with there separate --begin cert --- end cert --- blocks. I have tried to split it by using the additional chain field but to no avail.

RMerl, 2016-12-02:
Ask them which of the two certificates is the intermediate chain certificate, and put that one in the Extra Chain Certificate field.

siborg, 2016-12-02:
There are three in total. So I suspect there is the actual CA cert, intermediate CA cert and the actual cert. I've tried various combinations but nothing I put in the extra chain field gets saved

siborg, 2016-12-02:
I have received the reply from the VPN company. The actual certificate is 2048-bit but the whole intermediate chain takes it over the character limit per field.  I have been given just intermediate chain by the VPN provider but the config does not get saved when I hit apply. Only the CA field is saved. 

Anything I put in the Extra Chain Certificates field does not get saved.