Asuswrt-Merlin/asuswrt-merlin.legacy #1134: ipv6 passthrough and NAT Acceleration causes cyclic reboot

YuriyKischenko, 2016-12-06:
Enabling ipv6 passthrough and NAT Acceleration causes cyclic reboot in the latest Asuswrt 380.63_2 on rt-ac66u.
No issues with the latest firmware from Asus

RMerl, 2016-12-07:
Anything related to NAT acceleration is outside of my control.  If there was a bug that was fixed later on by Asus we will have to wait for new source code to become available for all models.

closed 2016-12-07 by RMerl

onetwobr, 2017-03-17:
I´m having the same issue on Asuswrt 380.65_2 in my AC66U too and this is not the first time I have faced exactly the same problem.

If with Asus firmware the IPv6 Passthrough works fine but overall performance on local network is poor, I'm starting to think about in possibility of a router bug. 

@YuriyKischenko Did you find any old version of Merlin where the problem didn´t occur?